particle-iot-archived/particle-dev #211: "Compiler timed out or encountered an error" on build target 0.6.2

rvess, 2017-06-15:
I noticed recently that if I had an error in my code, I would get the "Compiler timed out or encountered an error" message. I discovered that if I changed the build target, errors were again reported.
The following build targets reported errors ok: 
0.6.1-rc.1, 0.6.0-rc.2, 0.6.0-rc.1, 0.5.3, 0.5.3-rc.3, 0.5.3-rc.2, 0.5.3-rc.1, 0.5.2, 0.5.1, 0.5.0, 0.4.9, 0.4.7, 0.4.6, 0.4.5, 0.4.4, 0.4.3, 0.4.1
The rest result in the Compiler timed out message!

I even went as far as creating a new Particle account, installing Particle Dev on a different computer & logging in with the new account. This yielded exactly the same results as before!

I also checked with a friend and he is experiencing the same problem. 

This code is all it takes to reproduce the problem:

int test   // no semicolon so we generate an error!

void setup() {
}

void loop() {
}

Windows 10 (Version 1703, OS Build 15063.296)
Particle Dev (Version 1.8.0)
Photon

mjgorman, 2017-07-10:
I have more info to add here. I had multiple projects in my project tree. I added a project folder which contained multiple sub project folders with their own src. This caused the error. When I removed the parent project folder and only added the single project folder i'm working on to to the file browser is all works okay.

suda, 2017-07-20:
This bug has been fixed in the `particle-dev` package itself but the standalone Particle Dev App hasn't been updated yet. We're going to do it this sprint!

closed 2017-07-20 by suda

rvess, 2017-07-21:
Ok, thanks for the update.

suda, 2017-08-10:
@rvess The standalone Particle Dev App 1.18.0 ( https://github.com/spark/particle-dev-app/releases/tag/v1.18.0 ) app has been released :)