particle-iot-archived/particle-dev #212: Uncaught TypeError: Cannot read property 'apiClient' of undefined

shusain, 2017-06-16:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.18.0 x64
**Electron**: 1.3.15
**OS**: Ubuntu 16.04.2
**Thrown From**: [particle-dev](https://github.com/spark/particle-dev) package 0.1.22

### Stack Trace

Uncaught TypeError: Cannot read property 'apiClient' of undefined

```
At /home/shaun/.atom/packages/particle-dev/lib/views/select-core-view.coffee:57

TypeError: Cannot read property 'apiClient' of undefined
    at SelectCoreView.module.exports.SelectCoreView.loadCores (/packages/particle-dev/lib/views/select-core-view.coffee:57:47)
    at SelectCoreView.module.exports.SelectCoreView.show (/packages/particle-dev/lib/views/select-core-view.coffee:25:6)
    at SelectCoreView.show (/packages/particle-dev/lib/views/select-core-view.coffee:1:1)
    at /packages/particle-dev/lib/main.coffee:591:21
    at Object.loginRequired (/packages/particle-dev/lib/main.coffee:323:5)
    at Object.selectCore (/packages/particle-dev/lib/main.coffee:585:30)
    at /packages/particle-dev/lib/main.coffee:82:55)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/usr/share/atom/resources/app/src/command-registry.js:265:35)
    at CommandRegistry.handleCommandEvent (/usr/share/atom/resources/app/src/command-registry.js:3:65)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/usr/share/atom/resources/app/src/command-registry.js:166:25)
    at StatusBarView.module.exports.StatusBarView.selectCore (/packages/particle-dev/lib/views/status-bar-view.coffee:74:19)
    at /packages/particle-dev/node_modules/space-pen/lib/space-pen.js:220:36)
    at HTMLAnchorElement.dispatch (/packages/particle-dev/node_modules/space-pen/node_modules/jquery/dist/jquery.js:4435:9)
    at HTMLAnchorElement.elemData.handle (/packages/particle-dev/node_modules/space-pen/node_modules/jquery/dist/jquery.js:4121:28)
```

### Commands

```
     -4:54.8.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always)
     -4:54.7.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always)
     -4:54.7.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always)
  2x -4:43.4.0 core:backspace (input.hidden-input)
     -4:40 core:copy (input.hidden-input)
     -4:37.9.0 core:paste (input.hidden-input)
  4x -4:34.9.0 core:confirm (input.hidden-input)
     -3:49.6.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-light-syntax.theme-atom-light-ui)
     -3:49.6.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-light-syntax.theme-atom-light-ui)
     -3:40.9.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-light-syntax.theme-atom-light-ui)
     -3:40.9.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-light-syntax.theme-atom-light-ui)
     -0:15.1.0 particle-dev:select-device (atom-workspace.workspace.scrollbars-visible-always.theme-one-light-syntax.theme-atom-light-ui)
```

### Non-Core Packages

```
console-panel 0.2.1 
particle-dev 0.1.22 
particle-dev-cloud-functions 0.1.0 
particle-dev-cloud-variables 0.1.0 
particle-dev-local-compiler 0.1.0 
particle-dev-profiles 0.1.15 
tool-bar 1.1.0 
```
Attempting to just get things installed for compiling locally since I'm having some troubles with the linking on the web ide and wanted to see if local compiling helped but ran into a couple of errors, this one and the querySelector issue.

suda, 2017-07-06:
Hi Shaun, sorry for keeping you waiting. Is this error happens every time you try to select a device after opening a new Atom window? Do you see target platform and current build target selectors in the status bar?

sohil4932, 2017-07-18:
Hi having same issue after updating new package. Let me know how to resolve this.

suda, 2017-07-18:
@sohil4932 are you using standalone Particle Dev App or Atom?

sohil4932, 2017-07-18:
Standalone Particle Dev App. I downloaded it for Mac. I was using Particle Dev since long, but yesterday I updated a couple of Particle packages and after that, it stops working and giving me this error.

suda, 2017-07-20:
@sohil4932 can you check if you have any `particle-dev` packages installed as community ones? You can see this in Settings -> Packages -> Community packages

shusain, 2017-07-21:
Hi sorry I haven't had time to participate in this right now I ended up getting the build tools working on the command line and used that, I still had whatever problem with the linker when I was trying to divvy up my project into different cpp and .h files but it's been a long time since I really worked with C++ so likely something silly I did or something in the makefile that wasn't working out.  Seemed I could get it to compile everything but during the linking it failed to find things defined in files outside the application.cpp (since I was just toying with this I decided to just dump all the code in there).

Hopefully you all can get this issue sorted out though I did have to jump through some hoops to get compiling at the command line (think mostly just needed gcc arm build tools), good luck!