particle-iot-archived/particle-dev #214: Uncaught TypeError: path must be a string

zaprime, 2017-07-04:
I'm having a problem with Particle Dev when I try to compile in cloud

[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.8.0
**System**: Microsoft Windows 10 Pro
**Thrown From**: [particle-dev](https://github.com/spark/particle-dev) package, v0.1.20

### Stack Trace

Uncaught TypeError: path must be a string

```
At fs.js:584

TypeError: path must be a string
    at TypeError (native)
    at Object.fs.openSync (fs.js:584:18)
    at Object.module.(anonymous function) [as openSync] (ATOM_SHELL_ASAR.js:159:20)
    at Object.fs.readFileSync (fs.js:431:33)
    at Object.fs.readFileSync (ATOM_SHELL_ASAR.js:427:29)
    at C:\Users\Davidm\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:1030:40
    at Object.module.exports.minBuildTargetRequired (C:\Users\Davidm\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:498:14)
    at C:\Users\Davidm\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:941:24
    at Object.module.exports.loginRequired (C:\Users\Davidm\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:450:14)
    at Object.module.exports.compileCloud (C:\Users\Davidm\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:939:19)
    at atom-workspace.<anonymous> (C:\Users\Davidm\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:146:24)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (C:\Users\Davidm\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (C:\Users\Davidm\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (C:\Users\Davidm\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:160:19)
    at executeCallback (C:\Users\Davidm\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\tool-bar\lib\tool-bar-button-view.js:123:19)
    at ToolBarButtonView._onClick (C:\Users\Davidm\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\tool-bar\lib\tool-bar-button-view.js:82:9)
```

### Commands

```
     -3:28.1.0 console:toggle (ul.list-inline.tab-bar.inset-panel)
     -3:20.2.0 package-generator:generate-package (ul.list-inline.tab-bar.inset-panel)
     -3:17.4.0 editor:consolidate-selections (atom-text-editor.editor.mini.is-focused)
     -3:17.4.0 core:cancel (atom-text-editor.editor.mini.is-focused)
     -3:07.9.0 particle-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -2:42.1.0 particle-dev:compile-cloud (ul.list-inline.tab-bar.inset-panel)
     -2:42 window:save-all (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -2:42 particle-dev:update-compile-status (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
  6x -2:38.4.0 particle-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:05.6.0 particle-dev:compile-cloud (atom-text-editor.editor)
     -0:05.6.0 window:save-all (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:05.6.0 particle-dev:update-compile-status (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
```

### Config

```json
{
  "core": {
    "themes": [
      "one-dark-ui",
      "monokai"
    ]
  }
}
```

### Installed Packages

```coffee
# User
No installed packages

# Dev
No dev packages
```

[image]

I've checked my file names and the code runs fine on the online compiler so I'm at a loss as to what the problem is?

suda, 2017-07-06:
This bug has been fixed in the `particle-dev` package itself but the standalone Particle Dev App hasn't been updated yet. We're going to do it this sprint!

closed 2017-07-06 by suda

suda, 2017-08-10:
The standalone Particle Dev App 1.18.0 ( https://github.com/spark/particle-dev-app/releases/tag/v1.18.0 ) app has been released :)