particle-iot-archived/particle-dev #224: Package.json error

mynameishamish, 2017-09-08:
I'm getting this error when trying to use particle dev with my Redbear, I'm teaching a class using this board and Particle dev in Atom, and I'm getting this error on about 10% of my students machines when I try to open the Serial Port.

Can any one point me to a solution?

[image]

suda, 2017-09-11:
Hi Hamish! If you rename `~/.particledev` directory to something else (but not remove it) and start the Desktop IDE again, does it makes the error disappear?

mynameishamish, 2017-09-11:
That fixed it, thanks!

closed 2017-09-11 by mynameishamish

suda, 2017-09-12:
That's great! Could you zip the contents of it and send it to me at wojtek@particle.io ?