particle-iot-archived/particle-dev #225: Compile Error: "This project is only compatible with Particle system firmware v0.5.3 or later"

lpercifield, 2017-09-21:
Hello all,
I'm getting an error that prevents me from compiling code for the Redbear Duo board. After I initially start the Particle Dev IDE, I can compile my code targeting Photon without issue. If I target Redbear Duo, it will compile only once, then throw the error:
[image]
If I restart the IDE and compile for Photon then for Redbear Duo, it will again work, but only once.

[image]

Happy to provide more specifics if needed.

Thanks, and keep up the good work

suda, 2017-09-28:
Hi! This check is triggered when you're using libraries in Dev. It should be performed only for Particle devices and not compounds. We'll fix it asap

knud, 2017-11-15:
Hmmm, this seems still to be a problem, at least for me. But perhaps it's a bit different because what I do to get around it is to select a Photon as the device, but the Duo as the target. Compiling works in that case. Then I change devices back to my Duo and flash. This is all via the desktop IDE.

Any idea what's going on? Same problem as above?

maiiku, 2018-01-05:
@suda Any updates on this? I just tried to migrate from could particle IDE to desktop particle IDE due to lack of git support by cloud particle IDE. I found info that if I need to use git I need to switch to desktop IDE. So I manually copy-pasted all the files into the project, and... I get this error, and @knud's workaround doesn't work for me.

suda, 2018-01-05:
Unfortunately, there are no updates yet but I'll do my best to tackle it this sprint!

maiiku, 2018-01-08:
@suda Ok, please keep me posted :) It would be very nice if I could use your desktop IDE :)

closed 2018-01-08 by suda

suda, 2018-01-08:
We've just released `particle-dev` package with a fix for this issue. Note: this is only available if you're installed Particle Dev as [an Atom package](https://github.com/particle-iot/particle-dev#github-atom). We're working on updating the standalone package as well.

maiiku, 2018-01-11:
@suda I can confirm it is indeed working with the particle-dev package (I installed that package manually). Thanks for getting that fix out 👍 

[image]

commit 688f88e9 (2018-01-08): Fix #225