particle-iot-archived/particle-dev #240: Compile Error: "This project is only compatible with Particle system firmware v0.5.3 or later"

nayan27, 2018-02-20:
I have exact same problem as issue #225 but mine occurs with Raspberry Pi.

suda, 2018-02-20:
Hi @nayan27, unfortunately, this is a different issue as the Raspberry Pi firmware is, in fact < 0.5.3 and therefore not supported by the new library system. To make it work our fw should be updated but as we're in the process of [development of Particle Mesh](https://www.particle.io/mesh/) we can't assign resources for this. I'm very sorry for this inconvenience.

closed 2018-02-20 by suda

nayan27, 2018-02-20:
Does that mean i won't be able to develop firmwares with Particle Dev ? Can i use a earlier version ?
Our project is becoming large now to be working in the WebIDE.
Thank you,

suda, 2018-02-21:
You should be able to use CLI/Dev with Raspberry Pi but won't be able to use our library system. You could still manually copy the library files and compile the whole project. Try removing `project.properties` file and compile it in Dev.

nayan27, 2018-02-21:
Will give this a try. Though what do you mean by compiling in Dev ?

suda, 2018-02-22:
Oh I meant in Desktop IDE (AKA Particle Dev). Sorry for not being clear

nayan27, 2018-02-22:
Will give that a try thank you !