particle-iot-archived/particle-dev #241: Error: invalid_client

benjaminkraus, 2018-02-21:
I just installed the Particle Desktop IDE using a fresh install from the installer on the website (Windows 10 x64). https://docs.particle.io/guide/tools-and-features/dev/

When I first start using the Particle Desktop IDE, everything works fine. The first time I click the "Compile in the cloud and show errors if any" or the "Compile in the cloud and upload code using cloud" buttons they work successfully. However, after the first click something breaks, and those buttons (and most of the other Particle specific buttons) stop working. The only solution I have found is to close the Particle Desktop IDE and restart it.

I checked the developer console, and every ~30 seconds I see the following error:

Unhandled promise rejection Rejected with error: Error: invalid_client
    at Spark.normalizeErr (C:\Users\bkraus\AppData\Local\particledev\app-1.19.0\resources\app\node_modules\spark\lib\spark.js:117)
    at Spark.<anonymous> (C:\Users\bkraus\AppData\Local\particledev\app-1.19.0\resources\app\node_modules\spark\lib\spark.js:151)
    at Request.self.callback (C:\Users\bkraus\AppData\Local\particledev\app-1.19.0\resources\app\node_modules\request\request.js:188)
    at emitTwo (events.js:106)
    at Request.emit (events.js:191)
    at Request.<anonymous> (C:\Users\bkraus\AppData\Local\particledev\app-1.19.0\resources\app\node_modules\request\request.js:1171)
    at emitOne (events.js:96)
    at Request.emit (events.js:188)
    at IncomingMessage.<anonymous> (C:\Users\bkraus\AppData\Local\particledev\app-1.19.0\resources\app\node_modules\request\request.js:1091)
    at Object.onceWrapper (events.js:290)
    at emitNone (events.js:91)
    at IncomingMessage.emit (events.js:185)
    at endReadableNT (_stream_readable.js:974)
    at _combinedTickCallback (internal/process/next_tick.js:74)
    at process._tickCallback (internal/process/next_tick.js:98)

benjaminkraus, 2018-02-21:
[console.log](https://github.com/particle-iot/particle-dev/files/1742453/console.log)

suda, 2018-02-21:
Hi Benjamin, thank you for this detailed report! Could you try logging out and in again? This might be caused by an expired token

benjaminkraus, 2018-02-21:
I logged out and back in again. It did not appear to have any effect. In addition, this was a clean installation yesterday, so I logged in for the first time (via the Desktop IDE) yesterday. Also, restarting the IDE gets me back to a working state (I do not have to log in again).

Here is some additional details which may be helpful:

When I click "Compile in cloud and upload code using cloud" it will generate a file called "photon_0.6.3_firmware_1519225767139.bin" in the project directory, but does not actually upload the file. This is the same behavior as if I click "Compile in cloud and show errors if any". Once this is done, clicking any buttons (including the log-out button, the button to upload the code, the button to compile, or the button to select a different device) has no effect. Checking the console shows the error I posed before.

At this point, if I restart that IDE, I have to reselect my device, but then I can click the "Compile in cloud and upload code using cloud". As far as I can tell, it does not actually recompile. It sees the existing "photon_0.6.3_firmware_1519225767139.bin" and pushes it to the Particle Photon. The upload is usually successful, and it deletes the *.bin file from the directory. Once this operation is complete however, we are back into the broken state where clicking any buttons has no effect and errors are showing up in the console.

If it is related to my authentication token, then perhaps any attempt at communicating with the cloud (either compiling or uploading) is somehow corrupting the token? But restarting the IDE fixes the corrupted token? Not sure if that makes sense.